OllisGit/OctoPrint-DeleteAfterPrint #1: Deleted file should also be removed from state

BerndJM, 2018-04-20:
Just tested your plugin, works very well.
But removing the file from the entry under "state" would be nice to avoid confusion and unnecessary disconnect if one hits the print button again.

OllisGit, 2018-04-20:
Yes, of course ;-) And my tests were succesfull. The ProgressBar is switched to 0 and every other values are set to '-'.

hmmm...maybe I need more tests.

OllisGit, 2018-04-21:
@BerndJM Do you use the Print-History Plugin? It could be a problem with the `MetadataStatisticsUpdated`.
It looks like there is an "event-ordering" problem:
1. Event: Print done -> deleteFile
2. Event: MetaDataStatisticUpdate -> BOOOMM, file is gone!

I will investigate this problem further.

OllisGit, 2018-04-21:
The new Release 1.0.2 is out. Please test the bugfix.
Thx, in advance
Olli

BerndJM, 2018-04-21:
O.k. did some quick tests.
Seems the bugs are fixed, congratulation!

What I noticed (but I think thats an issue for the other plugin?):
your "DisplayLayerProgress" didn't take notice of deleted files (no matter if manual or automaticly deleted).

OllisGit, 2018-04-21:
Thanks for testing.
I created a new [bug](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/26) in my other repo.

The fix is also already implemented in Version 1.2.4

BR
Olli

closed 2018-04-22 by OllisGit