OllisGit/OctoPrint-DeleteAfterPrint #2: Feature Enhancement

oxgon, 2018-04-21:
add auto deletion based on time. 
Waiting days or weeks till deletion would be great!

OllisGit, 2018-04-22:
Looks like a good idea. I will check if it's possible to implement.
BR
Olli

oxgon, 2018-04-23:
wow, that was fast! I love this so far; it has saved me a lot of work, I print a lot and always hated going through my queue and spamming the delete button. Keep up the excellent work.

OllisGit, 2018-04-24:
@oxgon After installation and reloading the page, are all old files automaticaly deleted?
Because there is issue #3.
So it is really importend to know the behaviour after first installation. I couldn't reproduced this issue.
Please so kind and give me a fast answer!

Thx, in advance
Olli

oxgon, 2018-04-24:
@OllisGit I currently don't have any in there that are past the day mark.

closed 2019-01-11 by OllisGit