OllisGit/OctoPrint-DeleteAfterPrint #3: Allow users to choose behavior of old files, or add option to cancel deletion

Patronics, 2018-04-24:
My understanding of this plugin was that it offered a checkbox to enable deletion of completed prints on a per-print basis, but when I installed it I was surprised to find all of my old files were deleted! 

This behavior was not clearly explained, and I was planning to re-use some of those files. I had assumed it would only apply the auto deletion to new prints where you checked the "delete after print finishes", and not retroactively delete all prints already completed.

At an absolute minimum this needs to clearly and strongly be communicated to the user before installation of the plugin, and it really should not behave this way other than perhaps a "delete all old prints" button in settings. I'm sure I'm not the only person adversely affected by this behavior, and probably many don't have an easy way to restore all their deleted files.

OllisGit, 2018-04-24:
This shouldn't happend! I am really sorry!!

The plugin supports two features: a checkbox to activate deletion after print and (new in Version V1.1.0) delete files after predefined days.
The description of "day deletion" is not shown in the [Plugin-Repo page](https://plugins.octoprint.org/plugins/DeleteAfterPrint/), because the "Pull request" is pending for a couple of days. 
It is always a better idea to look at the github-page from the plugin, because here the author could change/provide informations much faster.

After installation the default daysLimit is 0 (turned off deletion after days).
**Did you changed the days limit in the plugin settings?** This importend to know for me, maybe there is something wrong with the "default settings behaviour".

BR
Olli

Patronics, 2018-04-24:
I did change the “delete after days” option, but I interpreted it as deleting the files that I checked the “delete after print” after keeping them a few days after the print finished, rather than deleting everything more than a few days old. That feature would be useful for me, keeping the files a few days after the successful print. 

And the current feature at least needs a better clarification of how it currently behaves, and ideally an “are you sure?” (stating that it will delete everything older than that immediately) pop-up when first enabling “delete after days”.

OllisGit, 2018-04-24:
Yes, your are right!
I will change it like this:
- remove the "daysLimit" definition from Settings-Page -> no _automatic_ deletion after days
- move the "daysLimit" field to sidebar
- add button in sidebar to manuelly execute the days-clean up [maybe with a confirmation-popup that shows all affected files]

what do you think?

oxgon, 2018-04-24:
@OllisGit  that sounds like a great addition!

OllisGit, 2020-03-01:
Sorry to say, but I don't change the settings like in 24.April 2018 mentioned. Makes to much effort.

closed 2020-03-01 by OllisGit