OllisGit/OctoPrint-DeleteAfterPrint #8: Add option to move the files into a specified folder instead of deleting.

Oxmaster, 2019-08-17:
I think this plugin could use an option to move files to a "trash" folder. So far I've been using the default option with octoprint to hide completed prints, but this is session specific, so if I log in with different device, this setting resets. All the rules would still be the same as the deletion option, but instead it would move the file instead of removing it. I don't know python (if I knew I would do it myself), but it should be an easy addition.

OllisGit, 2019-08-18:
@Oxmaster  sound like a good idea.
I will check if it is done with less effort.

OllisGit, 2019-08-31:
@Oxmaster the feature is implemented in newest version 1.5.0. 
Btw. It was not easy!!! I need to change a lot of stuff...several UIs, logik and testing, testing, testing.

Please test it and give me a feedback.
BR
Olli

OllisGit, 2019-11-02:
Closed: No feedback, so far!

closed 2019-11-02 by OllisGit