OllisGit/OctoPrint-DeleteAfterPrint #13: Additionsl Functional Idea

Brebinator, 2020-04-25:
Hi there, 

Thanks for writing this plugin. :) 

I thought perhaps based on the core outcomes of a job (Success, Failure, Stopped etc.) that perhaps some additional folders or a sendmail with an attached stub may be viable?

Just some thoughts and thanks. 
Brebs

OllisGit, 2020-05-03:
Hi @Brebinator !
hmmm...there were other plugins for notification.
Look into the plugin-repository:

https://plugins.octoprint.org/search/?q=notification

Brebinator, 2020-05-03:
Thx, will do.

https://plugins.octoprint.org/search/?q=notification

Kind Regards,
Ivan Brebner
Managing Director
digitalAtmosphere <http://www.digitalAtmosphere.com.au>

M: +61 423 658 569
W: +61 731 295 589
[Scene] <http://www.digitalatmosphere.com.au>
This message is supposed to be confidential. It may also be privileged or otherwise protected by work product immunity or other legal rules. But now that every phone has a camera, it seems that "privacy" is becoming more comical than ever. At dA we consider communciation to be precise and secure, so If you have received it by mistake, please let us know by e-mail reply and delete it from your system; you may not copy this message or disclose its contents to anyone. The integrity and security of this message cannot be guaranteed on the Internet.

closed 2020-05-03 by OllisGit