OllisGit/OctoPrint-DeleteAfterPrint #15: FEATURE: Do not show files deleted

AJErazzor, 2020-08-26:
Is it possible to add a feature to not show files deleted pop up, or they auto disappear in the UI?

OllisGit, 2020-08-26:
Hi @AJErazzor,

it's implemented in the latest Version 1.8.0.

BR
Olli

AJErazzor, 2020-08-30:
> Hi @AJErazzor,
> 
> it's implemented in the latest Version 1.8.0.
> 
> BR
> Olli

Hi Tested, checked hide after some time - the > X buttons are gonne till hover but the popup doesnt go away
[image]

OllisGit, 2020-08-30:
SORRY! Now it is fixed in the latest version 1.8.1

AJErazzor, 2020-09-03:
No popups visible but now Im not certain there ever is - is there a log that can show what has and hasnt been deleted to confirm?

AJErazzor, 2020-09-10:
I did manage to see a pop up of a file deleted and then disappear as expected.  Is it worth adding a new request to have a list view of the historically deleted jobs?  May not be necessary if using other of your fine plugins?

closed 2020-09-10 by AJErazzor