OllisGit/OctoPrint-DeleteAfterPrint #16: ReadOnly can delete file

billporte, 2020-11-02:
Hello,
It seems that Read Only Users can change the status of each parameters of deleteafterprint plugin.
but perhaps I misunderstand something.

OllisGit, 2021-01-11:
Hi @billporte ,
currently there is no permission-check implemented.

I marked it as "enhancement", because it is a good idea to implement such feature, but the Permission-Concept of OP was changed in Version 1.4, so I need to drop support for OP 1.3 of this plugin.

Which version of OP do you use (latest 1.5) ?

billporte, 2021-01-12:
Hi OlliGit,

Good news for moving it to enhancement
I stay permanently in the latest stable versions of octoprint or plugin

OllisGit, 2021-05-15:
Ho @billporte

I look into this topic with OP Version 1.6.1. It looks like this:
[image]
Yes, the checkboxes could be changed, but the user is not able to print.
So, I am a little bit confused what permission-check should assign to my plugin?
Any suggestion?

BR
Olli