OllisGit/OctoPrint-DeleteAfterPrint #17: Stopped working. TypeError: _historyFilterFunction() takes exactly 3 arguments (2 given)

mianos, 2020-12-26:
INFO - OctoPrint 1.5.2
|  DeleteAfterPrint Plugin (1.8.1) = /home/mianos/octo/lib/python2.7/site-packages/octoprint_DeleteAfterPrint

Crashes as follows. As a python developer I should go a bit further and fix it but it's not that serious until I end up with a large print list. :)

```python
2020-12-25 21:14:55,272 - octoprint.plugin - ERROR - Error while calling plugin DeleteAfterPrint
Traceback (most recent call last):
  File "/home/rfo/octo/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/rfo/octo/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/rfo/octo/lib/python2.7/site-packages/octoprint_DeleteAfterPrint/__init__.py", line 92, in on_event
    allFiles = self._file_manager.list_files(filter=self._historyFilterFunction)
  File "/home/rfo/octo/lib/python2.7/site-packages/octoprint/filemanager/__init__.py", line 693, in list_files
    force_refresh=force_refresh,
  File "/home/rfo/octo/lib/python2.7/site-packages/octoprint/filemanager/storage.py", line 682, in list_files
    result = apply_filter(result, filter)
  File "/home/rfo/octo/lib/python2.7/site-packages/octoprint/filemanager/storage.py", line 666, in apply_filter
    if filter_func(node) or node["type"] == "folder":
TypeError: _historyFilterFunction() takes exactly 3 arguments (2 given)
```

OllisGit, 2021-01-11:
Hi @mianos ,
please test the new release 1.9.0 and give feedback.

Thx, in advance
Olli

mianos, 2021-01-11:
Working now. My file list is all empty from no prints for a few days.

closed 2021-01-11 by mianos