OllisGit/OctoPrint-DeleteAfterPrint #19: Messes with TSD

jagr200, 2021-02-08:
So I love this plugin. I print hundreds of files a week and them being removed automatically is awesome. But when I use this and use The Spaghetti Detective it makes it so I don't get a print completed email from TSD until a start a new print. Is there anything that can be done to make it delete the file right away so Octoprint knows the print is complete and TSD will send the email?

OllisGit, 2021-02-13:
Hi @jagr200,

I looked into the TSD sourcecode,  but could not found how they decide to send/or not send an email.
My plugin just using the normal OP event-hooks, maybe you should raise a  ticket in the TSP-issue tracker.

OllisGit, 2021-05-15:
No feedback -> closed

closed 2021-05-15 by OllisGit