OllisGit/OctoPrint-DeleteAfterPrint #20: not working with octoprint 1.6.0

Speedy75, 2021-05-02:
its not working with latest octoprint, all errors are in log.

octoprint.log:
2021-05-02 11:19:43,686 - octoprint.plugin - ERROR - Error while calling plugin DeleteAfterPrint
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1941, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_DeleteAfterPrint/__init__.py", line 98, in on_event
    allFiles = self._file_manager.list_files(filter=self._historyFilterFunction)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/__init__.py", line 693, in list_files
    force_refresh=force_refresh,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/storage.py", line 666, in list_files
    result = apply_filter(result, filter)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/storage.py", line 650, in apply_filter
    if filter_func(node) or node["type"] == "folder":
TypeError: _historyFilterFunction() missing 1 required positional argument: 'entry_data'

OllisGit, 2021-05-15:
Hi @Speedy75,

new version 1.9.1 is out and should fix the issue. Please test and give feedback.

Thx, in advance
Olli

Speedy75, 2021-05-17:
thx, its working now

closed 2021-05-18 by OllisGit