OllisGit/OctoPrint-DeleteAfterPrint #22: Not An Issue, But a Feature Request

rebeltaz, 2021-06-17:
I love this plugin. Thank you so much for writing it. How much trouble would it be to add lock flag that could be toggled on a per-file basis? That way, if the flag is set for a particular file, regardless of the status of the DeleteAfterPrint flag, the particular file won't be deleted? 

I ask because there have been times that I have forgotten to turn Delete off and deleted files I'd rather keep on the Pi. This way, you could just leave Delete enabled and forget about it for the most part. 

Anyway... thanks again!

OllisGit, 2022-01-09:
Hi @rebeltaz 

new version 1.10.0 is out and now you can define filenames that should be excluded from delete/move action.

Please test and give a feedback.
Thx,
Olli

ttabi, 2023-04-14:
It would be nice if I could mark a file as "do not delete" from the Files panel on the home page.