OllisGit/OctoPrint-DeleteAfterPrint #23: Select folder to move to from sidebar dialog

ruedli, 2021-08-13:
Often I change the folder to move a file into after print. I then go into settings, find the plugin and change the folder.

It would be quit useful, to be able to select the folder to move to from a "pulldown". In theory, this pulldown could also feature the keyword [delete], so the delete option would not need a checkbox.

Currently the sidebar looks like this:

[image]

A dialog for selecting the folder to move/delete (similar when you "move" an item from the "Files" sidebar could look like this:

[image]

You could also add these two entries: 
[do nothing]
[delete]

OllisGit, 2022-01-09:
Hi @ruedli 

new version 1.10.0 is out and your suggested feature is implemented

Please test and give a feedback.
Thx,
Olli

ruedli, 2022-01-10:
well done! The user interface is as I had in mind and in my opinion easier to use. I have not tried it yet for a print.

Can you comment on the order of items in the pulldown? It does not seem to be alphabetical or creation time related, just wondering what the order is. Personally alphabetical makes most sense to me, but maybe this is difficult to realize.

ruedli, 2022-01-10:
I tested also adding a new folder and that folder being present in the pulldown / removed when the folder was removed. This worked as expected. 

As the user interface feature is now implemented, I close the issue. If there are issues on its implementation, e.g. how it works / does not work after a print, new issues should be initiated for that.

closed 2022-01-10 by ruedli