OllisGit/OctoPrint-DeleteAfterPrint #24: Missing 1 required positional argument: 'entry_data'

bruvv, 2022-01-08:
I receive the following error when I installed the addon.
```
2022-01-08 11:05:45,768 - octoprint.plugin - ERROR - Error while calling plugin DeleteAfterPrint
Traceback (most recent call last):
  File "/data/python/octoprint/lib/python3.9/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/data/python/octoprint/lib/python3.9/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/data/python/octoprint/lib/python3.9/site-packages/octoprint_DeleteAfterPrint/__init__.py", line 98, in on_event
    allFiles = self._file_manager.list_files(filter=self._historyFilterFunction)
  File "/data/python/octoprint/lib/python3.9/site-packages/octoprint/filemanager/__init__.py", line 688, in list_files
    result[dst] = self._storage_managers[dst].list_files(
  File "/data/python/octoprint/lib/python3.9/site-packages/octoprint/filemanager/storage.py", line 669, in list_files
    result = apply_filter(result, filter)
  File "/data/python/octoprint/lib/python3.9/site-packages/octoprint/filemanager/storage.py", line 653, in apply_filter
    if filter_func(node) or node["type"] == "folder":
TypeError: _historyFilterFunction() missing 1 required positional argument: 'entry_data'
```

OllisGit, 2022-01-08:
I could reproduce the issue with OP 1.7.
Will be fixed in next release.

OllisGit, 2022-01-09:
Hi @bruvv

new version 1.10.0 is out and fix the issue.

Please test and give a feedback.
Thx,
Olli

bruvv, 2022-01-09:
That is quick, thanks Olli! As soon as octoprint shows the update I will update and let you know

closed 2022-01-22 by bruvv